euslisp/EusLisp #477: llib/argparse.l usage??

Affonso-Gui, 2022-01-12:
@shmpwk 

With this program:
```lisp
(require :argparse "lib/llib/argparse.l")
(setq hoge (instance argparse:argument-parser :init))
(send hoge :add-argument "--foo" :default 0)
(print (send hoge :foo))
(exit)
```
We would expect `1` but we get `0` in the following:
```shell
$ eus argparse-test.l --foo=1
0
```

Affonso-Gui, 2022-01-12:
You are missing a `:parse-args` call:

```lisp
(require :argparse "lib/llib/argparse.l")
(setq hoge (instance argparse:argument-parser :init))
(send hoge :add-argument "--foo" :default 0)

;; HERE!
(send hoge :parse-args)

(print (send hoge :foo))
(exit)
```

```shell
$ eus argparse-test.l --foo=1
"1"
```
Note how the result defaults to a string; if you want the integer value you should add `:read t` and possibly a `:check #'integerp`:
```lisp
(send hoge :add-argument "--foo" :default 0 :read t :check #'integerp)
```

You can find more example usage at the test file, but you might need to decode some macros first:  https://github.com/euslisp/EusLisp/blob/master/test/argparse.l

**TODO:**
- warnings/errors when trying to access arguments without `:parse-args` ?
- better documentation (where to put it??)

closed 2022-01-12 by Affonso-Gui

shmpwk, 2022-01-12:
Got it, thank you for your quick response! We need `:parse-args` call like python argparse `parse_args()`.

> warnings/errors when trying to access arguments without :parse-args ?
I will try implementing it now.