euslisp/EusLisp #485: armhf builds on the ROS buildfarm are currently failing

clalancette, 2022-02-18:
Hi there,
     We recently noticed that builds of this package on the ROS buildfarm are failing; a recent example is https://build.ros.org/job/Mbin_ubhf_uBhf__euslisp__ubuntu_bionic_armhf__binary/374/, though it has been failing to build on both Noetic and Melodic for a while.
     The reason it has started to fail is because of a change we made on the ROS buildfarm.  Prior to ~ October 2021, we were building all armhf packages on AMD64 workers using qemu-arm-static binaries.  This worked, but was extremely slow.  Since October 2021, we have switched to using armhf containers on arm64 hosts in the cloud.  This is much, much faster.
     However, this package determines what architecture to build on based on the output of [uname -m](https://github.com/euslisp/EusLisp/blob/master/lisp/Makefile.LinuxARM#L39).  With an armhf container on an arm64 host, this returns `armv8`, and thus the wrong options are passed to the build.  This causes the assembler to fail later on.
     We haven't yet come up with a good way to solve this.  One way to fix it would be to change the detection in `Makefile.LinuxARM`, but we aren't quite sure what to replace it with.  Another way to solve this would be to patch the release in the release repository to use `dpkg-architecture` to pick the right kind of build.  A third way to fix it would be to make some changes to the buildfarm to still run things through `qemu-arm-static`, but on an arm64 host, which would hopefully be much faster than our previous solution.  We aren't quite sure the best way to proceed right now, but we wanted to open this issue to track the problem.

k-okada, 2022-02-24:
@clalancette thank you for your information. I have fixed on euslisp-release, and created PR on ros/rosdistro https://github.com/ros/rosdistro/pull/32296

https://github.com/tork-a/euslisp-release/blob/215a95593bc646dc6cee0a33b600b9343b4caba8/lisp/Makefile.LinuxARM#L40-L49

Since euslisp 9.27(https://github.com/ros/rosdistro/pull/32243) did not work on armhf , https://build.ros.org/job/Mbin_ubhf_uBhf__euslisp__ubuntu_bionic_armhf__binary/380/console, can we tray this new version?

clalancette, 2022-02-24:
Thanks for looking into it!  I will merge that PR in and give it a try.

closed 2022-02-24 by k-okada

clalancette, 2022-02-24:
@k-okada So it looks like that fixed the build of euslisp itself, but now downstream packages are failing.  One example is https://build.ros.org/job/Mbin_uB64__jskeus__ubuntu_bionic_amd64__binary/18/ .  Can you take a look?

k-okada, 2022-02-24:
@clalancette sorry, please try https://github.com/ros/rosdistro/pull/32298

clalancette, 2022-02-25:
Still not happy, unfortunately: https://build.ros.org/view/Mbin_ubhf_uBhf/job/Mbin_ubhf_uBhf__jskeus__ubuntu_bionic_armhf__binary/438/