euslisp/EusLisp #499: I encountered the bug after two rotations.

poyotamu000, 2023-03-28:
取り急ぎ

[image]

tkmtnt7000, 2023-03-28:
cc: @tkmtnt7000

tkmtnt7000, 2023-03-28:
Maybe correct.
```lisp
16.irteusgl$ load "geobody.l"
t
17.irteusgl$ (setq b (make-cube 10 10 10))
#<body #X563f4d6e2a70 (:cube 10.0 10.0 10.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
18.irteusgl$ (send b :rotate-vertices pi/2 :z)
#<body #X563f4d6e2a70 (:cube 10.0 10.0 10.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
19.irteusgl$ (send b :rotate-vertices pi/2 #f(0 0 1))
#<body #X563f4d6e2a70 (:cube 10.0 10.0 10.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>

```

k-okada, 2024-04-18:
closes via #500 #504

closed 2024-04-18 by k-okada