euslisp/EusLisp #505: unix::cdの引数にシンボルを与えるとエラーになる

Naoki-Hiraoka, 2024-02-13:
```lisp
$ eus
1.eus$ (unix::cd "/tmp")
t
2.eus$ (setq d "/tmp")
"/tmp"
3.eus$ (unix::cd d) ;; ERROR!
-2
4.eus$ (unix::cd (identity d))
t
5.eus$ (unix::chdir d)
t
```

unix::cdの引数にシンボルを与えるとエラーになる、という報告です。

代わりにjmanualに載っているunix::chdirを使うとエラーになりません。unix::cdはjmanualに載っていない関数なので、何か問題があるのかもしれません。

Affonso-Gui, 2024-02-13:
ダブルクォートを入れなくても良いように、敢えて評価しないようにしていると思います。

https://github.com/euslisp/EusLisp/blob/master/lisp/l/process.l#L12

```lisp
(cd /path/to/file)
```

Naoki-Hiraoka, 2024-02-14:
ありがとうございます。

> unix::cdはjmanualに載っていない関数なので、

と書きましたが、https://github.com/euslisp/EusLisp/blob/25a473c431559f62a18707f91ca5c1fa70feea5e/doc/jlatex/jsysfunc.tex#L785-L786 に載っていました。

https://github.com/euslisp/EusLisp/pull/506 でドキュメントを修正しました。

Naoki-Hiraoka, 2024-04-17:
https://github.com/euslisp/EusLisp/pull/506 がmergeされたためcloseします

closed 2024-04-17 by Naoki-Hiraoka