euslisp/EusLisp #519: How to change view distance of irtviewer?

sktometometo, 2025-02-04:
https://github.com/user-attachments/assets/bf8e82fe-f4be-4ab5-874b-250782b31ce9

```lisp
(setq a (make-cube 10 10 10))
(objects (list a)
```

This code shows a cube in a space with irtviewer. But since the viewer only shows within a specific distance, whole model cannot be visualized.
How can I change this parameter to see whole model?

pazeshun, 2025-02-04:
I'm not sure, but changing `gl::*perspective-near*` and `gl::*perspective-far*` may be effective.
Maybe you face this issue because those values are automatically changed:
https://github.com/euslisp/jskeus/blob/1.2.6/irteus/irtviewer.l#L219-L223
The default of those values are written here:
https://github.com/euslisp/jskeus/blob/1.2.6/irteus/irtviewer.l#L109-L110

sktometometo, 2025-02-04:
It works, thanks!

```lisp
(setq gl::*perspective-far* 50000.0)
(setq gl::*perspective-near* 0.0001)
```

closed 2025-02-04 by sktometometo