euslisp/EusLisp #526: How to detect nan value?

mqcmd196, 2025-05-01:
I want to process a function that returns `nan` .

I always write like

```lisp
(setq *c* (make-cube 0 0 0))
(eq (- *nan*) (send *c* :volume)) ;; t
```

However, I think it's a little weird having to write `(- *nan*)` instead of `*nan*` and ideally I want to write something like

```lisp
(if (send *c* :volume)
  ;; do something
```

Do you have any features?

k-okada, 2025-05-02:
This is very interesting, 
according to https://ja.wikipedia.org/wiki/NaN, NaN have signed bit and 
```
  float a = 0/0.0;
  printf("%f\n", a);
```
show `-nan` for normal GCC, but `send <body> :volume`  is calculated as the product of `:plane-distance` and `:area` 
https://github.com/euslisp/EusLisp/blob/f72322f1f784ee846a706e16cc5e5e0edfa4ee70/lisp/geo/geobody.l#L229-L230

https://github.com/euslisp/EusLisp/blob/f72322f1f784ee846a706e16cc5e5e0edfa4ee70/lisp/geo/geopack.l#L924-L927
Thus, `(send *c* :volume)` becomes `+NaN = (- *nan*)`.

you better to write isnan function to check if :volume is *nan*
```
(defun isnan(x)
  (or (equal x *nan*) (equal x (- *nan*))))

(defun isnan(x)
  (equal (logand x #x7fffffffffffffff) #x7ff8000000000001)) ;; this code not working for 32bit 
```

https://github.com/euslisp/EusLisp/blob/a58fd4f38f89922d54622386af5ed9eea148a515/lisp/geo/render.l#L244-L245

mqcmd196, 2025-05-04:
I see. It depends on the C compiler. And the reason why eus returns +NaN in x86 is the implementation of volume calculation. 

FYI

```c
#include <stdio.h>

int main(){
    float a = 0 / 0.0;
    printf("%f\n", a);
    return 0;
}
```

in x86 prints `-nan`

in aarch64 prints `nan` . 

```
 (defvar *nan*  0.0) 
 (setq *nan* (/ *nan* 0.0)) 
```

tries to keep the architecture compatibility?

k-okada, 2025-05-09:
Yes, C compiler outputs both +NaN and -NaN, so it is impossible to represetn NaN as a signle simbol in euslisp, thus why we should define the `isnan` function.

closed 2025-05-13 by mqcmd196